mantidproject/mantid #22242: Instrument view shows now a gap between left and back SANS detectors

reimundILL, 2018-04-03:
LoadEmptyInstrument D33:

Release 3.12:

[image]

Master:

[image]

Expected behavior:
- no zickzack in particular when detector defines rectangular pixels in xy plane
- there should not be a gap between the left and back detector (it suggests that the IDF is wrong!)

OwenArnold, 2018-04-03:
@reimundILL Thanks for bringing this to our attention. How urgent is the fix required in Master for this?

gvardany, 2018-04-03:
@OwenArnold It is not critical for D33, but maybe other rectangular detectors are also skewed.

reimundILL, 2018-04-03:
The instrument view was my IDF verification tool for D33. I trust it in terms of updating the IDF. Just let me know when I can have a look at it again. Thanks for your investigations.

commit 96a25bf2 (2018-04-10): review fix #22242
PR #22262 (2018-04-11): Corrected Rendering of Rectangular Detectors
Added correction to the rendering of rectangular detectors. Offsets and rotations were not handled correctly. This was highlighted by the D33 ILL instrument.

**To test:**

Ensure the D33 Instrument and other instruments which use `RectangularDetector` are rendered correctly. (LOQ, SANS2D etc).

Fixes #22242 . 

**Release Notes** 

*Internal fix does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

-  [... cut]
closed 2018-04-11 by martyngigg