mantidproject/mantid #22247: Test for multiple scattering

Mantid-Matthew, 2018-04-04:
Implement Grethe Jensen/John Barker’s approach for testing for higher order contributions to the scattering:
-  Set an adjustable target fraction of the scattering that must be accounted for (eg, F=0.95)
-  Determine the scattering power (the average number of times a neutron is scattered) tau* that corresponds to a probability of single scattering equal to F
-  Determine the actual scattering power of the sample by integration of I(q), call this tau
-  Raise a flag/message if tau > tau* specifiying: i) the number of scattering orders that must bee included to account for F, and ii) indicate how much the wavelength or sample thickness (or both) must be decreased for tau < tau*

smk78, 2020-07-16:
It would also be nice to automatically compute the wavelength overlap 'under the hood' and warn the user if it is bad. Though this would need implementing in a way that was mindful of reduction speed issues.

RichardHeenan, 2020-07-27:
Poor overlap of the different wavelengths may be caused by a number of issues such as one or more of: multiple scattering, Bragg scattering (or other strong wide angle diffraction), very low transmissions, mis-matched transmissions between sample and can, a weakly scattering sample with stronger incoherent and/or inelastic scatter, sample not properly centred in beam thus creating background issues.
Scientists need to develop some statistical measures for "poor overlap". An estimate of the size of multiple SANS scatter as outlined above, or by Monte-Carlo etc, could be flagged to the users when above a threshold. Someone needs to prototype some routines.

closed 2021-09-06 by stale[bot]