mantidproject/mantid #22254: Change Algorithm doc landing page to categories

NickDraper, 2018-04-05:
This is to change the landing page for documentation to be the base category list

it will also:

1. contain a link to the full alphabetical list on the first landing page.
1. neaten up the list handling when there are few items to list.

NickDraper, 2018-04-05:
So i didn't get any further,

It fails with this message
```
6>  AttributeError: 'str' object has no attribute 'name'
```

but I haven't been able to track it down yet

keeeto, 2018-04-06:
This is starting to come together - there is currently an issue in that subcategory pages are now taking on the over-arching category format. I'll look at this after lunch.

keeeto, 2018-04-06:
I think that some of the category names could be made more descriptive - eg "Crystal" and "Background" are terms that could be any number of things.

PR #24904 (2019-03-05): Change alg landing page to categories, not full index.

**Description of work.**
See changes in the to test section

**Report to:** nobody. 

**To test:**

1. Open the algorithms page from a variety of links (e.g. mantid home->documentation, mantidplot help page, anywhere else you can think of.)
1. Consider the changes below and check that they work (apart from perhaps the last one.)
1. Make sure that you test with the offline help as well as the generated html pages.
1. It would be worth testing on a non winodws build as there is a certain amount of path manipulation involved in this.

Change [... cut]
closed 2019-03-05 by SimonHeybrock