mantidproject/mantid #22421: Failed to initialize Python starting MantidPlot

jackhodkinson, 2018-05-13:
Launching MantidPlot (3.12.1) using the MantidPlot.app launcher on MacOS High Sierra (10.13.4) gives an error message "Failed to initialize Python. Please contact support."

Workspaces won't load and algorithms won't run.

Examining the Results Log after launching the program I see an error after the following:

```
Opening all libraries in /Applications/MantidPlot.app/plugins/

Traceback (most recent call last):
  File "<string>", line 23, in <module>
  File "/Applications/MantidPlot.app/Contents/MacOS/mantidplot.py", line 10, in <module>
    import pymantidplot
  File "/Applications/MantidPlot.app/Contents/MacOS/pymantidplot/__init__.py", line 18, in <module>
    import pymantidplot.proxies as proxies
  File "/Applications/MantidPlot.app/Contents/MacOS/pymantidplot/proxies.py", line 16, in <module>
    import mantid
  File "/Applications/MantidPlot.app/Contents/MacOS/mantid/__init__.py", line 78, in <module>
    import mantid.plots as plots
  File "/Applications/MantidPlot.app/Contents/MacOS/mantid/plots/__init__.py", line 26, in <module>
    import mantid.plots.plotfunctions
  File "/Applications/MantidPlot.app/Contents/MacOS/mantid/plots/plotfunctions.py", line 24, in <module>
    import matplotlib.dates as mdates
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/dates.py", line 119, in <module>
    from dateutil.rrule import (rrule, MO, TU, WE, TH, FR, SA, SU, YEARLY,
  File "/Users/Jack/Library/Python/2.7/lib/python/site-packages/dateutil/rrule.py", line 19, in <module>
    from six.moves import _thread, range
ImportError: cannot import name _thread
Error running init file "/Applications/MantidPlot.app/Contents/MacOS/mantidplotrc.py"
Sending https GET request to: https://api.github.com/repos/mantidproject/mantid/releases/latest
DownloadFile : https://api.github.com/repos/mantidproject/mantid/contents/instrument to file: /Users/Jack/.mantid/instrument/github.json
Sending https GET request to: https://api.github.com/repos/mantidproject/mantid/contents/instrument
Unable to find proxy.host in the properties file
Unable to find proxy.host in the properties file
NetworkProxyOSX. No system HTTP Proxy set!
NetworkProxyOSX. No system HTTP Proxy set!
Sending http POST request to: http://reports.mantidproject.org/api/usage
Answer from web: 201 Created
Answer from web: 200 OK
GitHub API 4999 of 5000 calls left. Resets at 2018-May-13 23:22:12 GMT
Current Mantid Version: 3.12.1
Most Recent Mantid Version: 3.12.1
Time to validate properties: 0.000004 seconds
Time for other input validation: 0.000000 seconds
Time for other initialization: 0.000017 seconds
Time to run exec: 0.599554 seconds

Answer from web: 304 Not Modified
Answer from web: 304 Not Modified
All instrument definitions up to date
Time to validate properties: 0.000012 seconds
Time for other input validation: 0.000000 seconds
Time for other initialization: 0.000034 seconds
Time to run exec: 0.721272 seconds

Previous initialization of Python failed, cannot retry.Determining ioRaw buffer size
Unable to find loadraw.readbuffer.size in the properties file
loadraw.readbuffer.size not found, setting to 200000
Determining ioRaw buffer size
Unable to find loadraw.readbuffer.size in the properties file
loadraw.readbuffer.size not found, setting to 200000
Determining ioRaw buffer size
Unable to find loadraw.readbuffer.size in the properties file
loadraw.readbuffer.size not found, setting to 200000
Determining ioRaw buffer size
Unable to find loadraw.readbuffer.size in the properties file
loadraw.readbuffer.size not found, setting to 200000
Determining ioRaw buffer size
Unable to find loadraw.readbuffer.size in the properties file
loadraw.readbuffer.size not found, setting to 200000
Determining ioRaw buffer size
Unable to find loadraw.readbuffer.size in the properties file
loadraw.readbuffer.size not found, setting to 200000

...
```

NickDraper, 2018-06-28:
Whenever we see a problem like this it is almost always a problem with your local installation.
Try reinstalling Mantid again.

jackhodkinson, 2018-06-28:
Hey Nick,

Thanks for your response.

From what I can tell the issue is with Mantid's python interpreter calling an outdated version of the python library six.moves which does not have "_thread".

I checked all my installations of python on my computer, Homebrew, Conda, and the default python that comes installed on the Mac, and I found the first two have the correct version of six.moves, but the default version of python on my computer does not. From what I can tell, MacOS prevents you from updating libraries in any /System/Library/Frameworks/Python.framework/ directory so I can't update it.

On the upside I have managed to install the mantid framework (without the gui) via Conda, but it would be nice to have the gui for some things. Please let me know if you have any further advice.

NickDraper, 2018-07-05:
Keith, can you or Sam see if you can help out.

keeeto, 2018-07-05:
Hi @jackhodkinson 

It looks like there is a conflict being caused during the loading of `dateutil`.

From the location: `/Users/Jack/Library/Python/2.7/lib/python/site-packages/dateutil/rrule.py` I assume that this is a package that you installed using `homebrew` or `pip`. Do you recall installing dateutil?

If you could try removing that package/taking it off the PYTHONPATH and then reloading MantidPlot we could diagnose the cause.

MantidPlot ships with its own version of Python and the various libraries required, so sometimes if there is another version on the system, with an incompatible.

jackhodkinson, 2018-07-05:
Thanks @keeeto! I did as you suggested and now it works!

I do not remember installing anything in that directory. My site packages are usually installed in other places with pip or homebrew, but it could be the case that these were installed from my earlier days before I learned how to use a package manager... Anyway, mantidplot started up as expected and runs properly after I removed the dateutil package from that directory.

Many thanks.

closed 2018-07-05 by jackhodkinson

keeeto, 2018-07-05:
Glad that worked @jackhodkinson !!