mantidproject/mantid #22442: Plus Algorithm not commutative on EventWorkspaces with different X values

MikeHart85, 2018-05-18:
`Plus` is a `CommutativeBinaryOperation`.

But it will produce different results when used on two EventWorkspaces that have different X values. The output workspace will inherit the X values of the LHS input workspace. The RHS input workspace's X values are ignored.

The [documentation](http://docs.mantidproject.org/nightly/algorithms/Plus-v1.html) for the Plus algorithm really doesn't say anything about this scenario. Several examples have comments saying `#X-values must be identical`, but none of the examples use EventWorkspaces.

It's not quite clear what _should_ happen to the X values in this scenario. Since they represent bin boundaries, being different implies the TOF of events are in different ranges. That means that the resulting workspace is very likely to have events that are outside of both LHS and RHS X values (individually).

One option might be to reset the X values completely using `IEventWorkspace::resetAllXToSingleBin()` that was introduced in #22425. If this approach is taken, the reset in `LoadLiveData::exec` might become redundant in "Add" mode.

Another option might be to require the X values to be the same for `Plus` to work at all. A downside here is that this would require manually rebinning both LHS and RHS to X values that work for both when you want to add two EventWorkspaces that don't match.

Related and similar algorithm, like `PlusMD`, `Minus`, etc, should also be looked at to make sure they are consistent.

AndreiSavici, 2018-05-18:
The only sensible option is to make one bin, unless bin boundaries are the same

NickDraper, 2018-06-28:
X bin boundaries have little real solid meaning for event workspaces, and are not even considered during the plus operation as all it will do is concatenate the event lists.  The X values are just the bin boundaries that the dynamic histogramming uses when displaying the events.  I'm now convinced that this is an issue that should be resolved.  Reopen if you strongly disagree.

NickDraper, 2018-06-28:
You could always have events outside the visible range, just call rebin again!

closed 2018-06-28 by NickDraper