mantidproject/mantid #22465: EstimateMuonAsymmetryFromCounts : First bin in data not included in normalization.

TWJubb, 2018-05-23:
### Expected behavior

In ``EstimateMuonAsymmetryFromCounts`` algorithm, I feed in a histogram with bin edges (0.0,0.1,0.2,...). The calculation of the asymmetry requires a summation of the counts from all of the bins between a start and end time. If I set my start time to be equal to the lowest bin edge (i.e. t_0 = 0.0), then I expect that the counts from the first bin (0.0-0.1) are included in the summation. Currently that is not the case

### Actual behavior

https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/MuonAsymmetryHelper.cpp#L87

If the first bin edge of dataX is equal to startX, the summation will not include the first bin. This is caused by the ```startIndexFromTime``` function

https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/MuonAsymmetryHelper.cpp#L106

which uses ```std::upper_bound```, which uses a 'greater than' rather than 'greater than or equal to' comparison.

AnthonyLim23, 2018-05-29:
http://www.cplusplus.com/reference/algorithm/for_each/

TWJubb, 2018-10-09:
I am reopening this issue

https://github.com/mantidproject/mantid/blob/master/Framework/Muon/src/MuonAsymmetryHelper.cpp#L124

The use of `std::lower_bound` does give the correct behavior, but then as in the line above the returned value is `upper + 1`; it should be `upper` to fix the issue.

AnthonyLim23, 2018-10-09:
the plus one is intentional. If you have bins from 1 to 10 in increments of 1. The if you give it an end bin of 9.5, it is better to round up to 10 then to exclude some data that the user was expecting.

TWJubb, 2018-10-09:
I think there is some confusion here. I agree with what you've said, but the code I'm referring to and linked to above is for the lower end of the range (not the upper end).

You will see that the data is actually being cut off from the lower bound; hence the user is losing data that they expected... 

So if e.g. I have

```
std::vector<double> xvals = { 0.0, 0.1, 0.2, 0.3, 0.4, 0.5 };
auto upper = std::lower_bound(xvals.begin(), xvals.end(), 0.0);
size_t i = std::distance(xvals.begin(), upper + 1);
```
then I return `i=1`, whereas I should expect to have `i=0`

AnthonyLim23, 2018-10-09:
this is subtle, but it is currently correct. Using the example above. If the first good data is at 0.5 then we do not want anything less than 0.5 as the data will be nonsense. Hence, in this case it is better to exclude from 0.5 to 1 as that bin contains some bad data. This is more to do with the nature of the experiment and the start/end behaviour is different.

TWJubb, 2018-10-09:
> If the first good data is at 0.5 then we do not want anything less than 0.5 as the data will be nonsense. Hence, in this case it is better to exclude from 0.5 to 1 as that bin contains some bad data.

If the first good data (i.e. the `xStart` as it is called in the code) is at 0.5 and we have the data

```
{0.1,0.2,0.3,0.4,0.5,0.6,0.7,0.8,0.9,1.0}
```

Then we expect the following data after cropping

```
{0.5,0.6,0.7,0.8,0.9,1.0}
```

No? As this (as you say) does **not** include anything less than 0.5. Currently we would have instead

```
{0.6,0.7,0.8,0.9,1.0}
```

Thus cutting away the bin `[0.5,0.6]`, which we should not be doing because the user has requested all data above `0.5`.

I agree with everything you have said; if we set `xStart =0.55` then we expect `{0.6,0.7,0.8,0.9,1.0}`. However if we sit **right on** the lower bin edge at `xStart = 0.500000`, then we should be including the bin. This is what the original issue is meant to address.

TWJubb, 2018-10-10:
After some further investigation; the problem lies around here

https://github.com/mantidproject/mantid/blob/master/Framework/Muon/src/MuonAsymmetryHelper.cpp#L94

with the use of `startIndexFromTime(xData, startX)` (see comment above https://github.com/mantidproject/mantid/issues/22465#issuecomment-428223954). This is incorrectly cropping the lower end of the input time range.

This **does not** affect the x-axis of the data, it only affects the calculation of the normalization `N0`. So it would seem the asymmetries are being normalized with an inconsistent value.

Referring to the algorithm's documentation 

I believe the fix is as per the original comment (value is `upper + 1`; it should be `upper` to fix the issue).

TWJubb, 2018-10-10:
Another question;

https://github.com/mantidproject/mantid/blob/master/Framework/Muon/src/MuonAsymmetryHelper.cpp#L106

Shouldn't this be 'kN <= iN' not `kN < iN`? Again we appear to be removing the last bin of the correct range; since `iN` already accounts for the truncation effect (e.g. if we request `0.55` the `iN` accounts for truncating to the bin which ends at `0.5`).

TWJubb, 2018-10-10:
[EstimateMuonCountsFromAsymmetry.pdf](https://github.com/mantidproject/mantid/files/2464870/EstimateMuonCountsFromAsymmetry.pdf)

I have summarized the main algebra for the calculation in the attached PDF; I have used the code as a reference, as well as the documentation. Couple of immediate issues : 

- What I've said about the unwanted removal of the first bin in the data is still true; I think this may be a bug.
- The documentation page for this algorithm contains equations which are confusing and inconsistent with what is done in the code. This could be fixed using the LaTeX equations from the document (if they are correct).
- In Eq.(1.1) of the attached PDF; the `x_i` which are used; should these be bin centers or bin edges? I believe they should be bin centers which is not what is currently done.

TWJubb, 2018-10-11:
For the sake of being concrete, I will take the following example;

```
x_data =  [0.0, 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8, 0.9, 1. ]
y_data =  [50,40,25,3,0,0,0,0,0,0]
```

i.e. a histogram with 10 bins. Now I am calculating the normalization N0, and the asymmetry by hand according to the formula given in the documentation. I get the following values (number of good frames is set to `10`);

```
N0 = 1.4685994619
asymmetry = [ 2.40460427,  1.85052209,  0.86454218, -0.7658354 , -1., -1., -1., -1., -1., -1.]
```

These match the algorithm via 

```
x_data = [0.0, 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8, 0.9, 1.]
y_data = [50, 40, 25, 3, 0, 0, 0, 0, 0, 0]
input = simpleapi.CreateWorkspace(x_data, y_data)
run = input.getRun()
run.addProperty("goodfrm", "10", "None", True)
output = simpleapi.EstimateMuonAsymmetryFromCounts(InputWorkspace=input, 
                                                    NormalizationTable=tab,
                                                    spectra=0, 
                                                    StartX=0.0, 
                                                    EndX=1.0)
print("Asymmetry x:  {}".format(output.readX(0)))
print("Asymmetry y:  {}".format(output.readY(0)))
```

**only** after making the two changes.

PR #23998 (2018-11-16): Muon asymmetry bins
The bins were not being excluded correctly in the muon algorithms. This PR fixes that.
**Report to:** [user name]/[nobody]. 

**To test:**
use the below script. For the range given in the asymmetry calculation (0.05 to 0.95) we expect the normalization to be unaltered (comment out line 9 = the spike in the data) for index = 1 or 9 (exclude first and last bins). For any other valid value of the index the normalization will be much larger. 

```
import numpy as np

x_data = np.linspace(0,1,num=11)
y_data= 1.0/np.exp(x_data/2.2)
#make hist data
y_data = y_data[:-1]

index = 1
 [... cut]
closed 2018-11-16 by soininen