mantidproject/mantid #22551: Crystal Field: let external code modify the hamiltonian.

mantid-roman, 2018-06-07:
Allow the crystal field functions to use external code (an algorithm?) provided by the user to modify the hamiltonian before diagonalising it.

AnthonyLim23, 2020-09-18:
@mducle could you explain in a bit more detail what this issue is describing? Is it still something that you want?

mducle, 2020-09-18:
I can't actually remember what this was about - and I can't find any emails about it either. I think it was related to something Petr Cemak wanted to add. Looking at the code that Roman wrote, it seems the idea is that the user can define a transformation of the Hamitlonian matrix after it is calculated from the CF parameters but before it is diagonalised. The user would have to define their own python function to modify the Hamiltonian and register it as a Mantid algorithm. Mantid would then apply this transformation each time before it diagonalises the Hamiltonian. 

I vaguely remember that this had something to do with including extra effects not calculated by the routines here but I can't remember what it was - I'm pretty sure it was something requested by Petr Cermak but he didn't follow up on it, and I forgot about it.

closed 2021-03-25 by stale[bot]