mantidproject/mantid #22553: Add numpy `dtype` exports to PropertyWithValue

samueljackson92, 2018-06-07:
The current exports of `PropertyWithValue`do not provide a way to export the type of the property. This leads to us having to write gross pieces of code such as the following:

```python
        if isinstance(prop, (FloatPropertyWithValue, FloatTimeSeriesProperty, FloatArrayProperty)):
            return "f"
        if isinstance(prop, (IntPropertyWithValue, Int32TimeSeriesProperty, Int64TimeSeriesProperty)):
            return "i"
        if isinstance(prop, (BoolPropertyWithValue, BoolTimeSeriesProperty)):
            return "b"
        if isinstance(prop, StringPropertyWithValue):
            return "S{}".format(len(prop.value))
        raise RuntimeError("Unrecognised property type: \"{}\". Please contact the development team with this message".
                           format(prop.type))
```

Instead I propose we expose a new function called `dtype` that returns a numpy friendly string format for the `dtype`. [See here](https://docs.scipy.org/doc/numpy-1.14.0/reference/generated/numpy.dtype.html) for the docs of the numpy function.

josephframsay, 2018-06-07:
If I may take liberties, from the point of view of #22554 there's one extra thing that will be needed for this change to be used in `ExportSampleLogsToHDF5` (and one thing that would be nice).

The necessary thing is that we'll also need to be able to get the dtype of a `TimeSeriesProperty`. In `ExportSampleLogsToHDF5` this is the same as if the log were a scalar (ie `IntTimeSeriesProperty` has dtype `i`, not array of `i` or anything like that).

The nice-to-have is that `ExportSampleLogsToHDF5` also has to check whether the log is a `TimeSeriesProperty`, which it currently does with another great big `if isinstance(blah ...)`. This might be beyond the scope of this issue, but I think it's another thing that would be generally useful?

@samueljackson92 thoughts?

samueljackson92, 2018-06-08:
@josephframsay Sounds like you want a `isTimeSeries()` method on the `Property` export. That seems like an entirely sensible suggestion.

PR #22644 (2018-07-26): Numpy dtype exports to property with value
**Description of work.**
Implemented a new function called `dtype()` which is exposed to the Python side that returns a numpy friendly string format for the `dtype`.

Documentation of the `numpy` function:
https://docs.scipy.org/doc/numpy-1.14.0/reference/generated/numpy.dtype.html

**Report to:** @samueljackson92  

**To test:**
Not much to test specifically - just run the Python unit tests.

`ctest -C Debug -R TimeSeriesProperty`

Fixes #22553. 

Does this update require release notes?
- [ ] Yes
- [x] No

---

#### Reviewer ####

Please comment on t [... cut]
closed 2018-07-26 by martyngigg