mantidproject/mantid #22555: Make it easier to construct TimeSeriesPropertys on the python side.

samueljackson92, 2018-06-07:
In python, if I want to create a `TimeSeriesProperty` but I don't know the type of the values I have to do the following:

```python
        if isinstance(log_value[0], int):
            prop = Int32TimeSeriesProperty(log_name)
        elif isinstance(log_value[0], float):
            prop = FloatTimeSeriesProperty(log_name)
        elif isinstance(log_value[0], str):
            prop = StringTimeSeriesProperty(log_name)
        elif isinstance(log_value[0], bool):
            prop = BoolTimeSeriesProperty(log_name)
        else:
            raise RuntimeError("Unsupported property type {}".format(type(log_value[0])))
```

Perhaps we can do this differently. I propose we have a python side exported `TimeSeriesProperty` constructor function which works like the following:

```python
log_name = 'test'
log_values = [2., 3., 4.]
prop = TimeSeriesProperty(log_name, log_values)
print(type(prop))
```
Which prints:
```bash
<type 'FloatTimeSeriesProperty'>
```

BhuvanBezawada, 2018-06-07:
So essentially we want to make constructor which would do the work in the background and it would do it more efficiently?

samueljackson92, 2018-06-08:
Not more efficient per se, just easier to use and more pythonic. It also won't *technically* be exporting a real constructor because there's no such thing as a `TimeSeriesProperty` on the python side, only there concrete implementations. But we can make it look & feel like that's what you're doing.