mantidproject/mantid #22556: Refactor MayersSampleCorrections

marshallmcdonnell, 2018-06-07:
Recently, there was a refactor of `MultipeScatteringCylinderAbsorption` into two separate algorithms, one for calculating the corrections,  [CalculateCarpenterSampleCorrection](http://docs.mantidproject.org/nightly/algorithms/CalculateCarpenterSampleCorrection-v1.html#algm-calculatecarpentersamplecorrection),  and another for calculating and applying the corrections, [CarpenterSampleCorrection](http://docs.mantidproject.org/nightly/algorithms/CarpenterSampleCorrection-v1.html#algm-carpentersamplecorrection).

This is to apply the same refactor to the [MayersSampleCorrection](http://docs.mantidproject.org/nightly/algorithms/MayersSampleCorrection-v1.html).

Also, in addition, the multiple scattering correction is in disagreement with the CarpenterSampleCorrection multiple scattering. The Mayers is suspect to have a bug do to the extent of the correction, shown below. Here, the absorption corrections for both are in almost perfect agreement but when applying the multiple scattering on the bottom plot, we see the MayersSampleCorrection algorithm subtracts more of the signal than expected for vanadium. The Carpenter multiple scattering correction is shown in the middle plot and on par with levels reported by previous literature.
[image]

Anders-Markvardsen, 2018-06-22:
Hi @marshallmcdonnell , One thing I don't understand is that this report https://github.com/mantidproject/documents/blob/master/Presentations/Multiple%20Scattering%20Presentation/Investigating%20multiple%20scattering%20with%20McStas.docx shows that the Mayers/Carpenter absorption corrections are different. This report includes python scripts. Would you be able to explain why the absorption comparison in this report is different from the above? Thanks

marshallmcdonnell, 2021-02-22:
WIP

closed 2022-02-26 by stale[bot]