mantidproject/mantid #22576: PDCalibration unable to gracefully handle error for FitPeaks

marshallmcdonnell, 2018-06-13:
**Original reporter:** Marshall McDonnell, SNS, ORNL (GitHub username: marshallmcdonnell)

### Expected behavior
If a peak is not found during the `FitPeaks` part of `PDCalibration` for a given region, there should be a mechanism to handle the exception gracefully (i.e. input that the pixel should be masked). 

### Actual behavior
`FitPeaks` will give the following error, causing `PDCalibration` stop:

```

### Steps to reproduce the behavior

### Platforms affected

peterfpeterson, 2018-06-19:
I cannot reproduce the error with the following script. What am I missing?
```python
LoadEventNexus(Filename='NOM_108448', OutputWorkspace='NOM_108448')
CompressEvents(InputWorkspace='NOM_108448', OutputWorkspace='NOM_108448')
dvalues = (0.3117, 0.3257, 0.3499, 0.4205, 0.4645, 0.4768, 0.4996, 0.5150,
           0.5441, 0.5642, 0.5947, 0.6307, .6866, .7283, .8185, .8920,
           1.0758, 1.2615, 2.0599)

PDCalibration(InputWorkspace="NOM_108448",
                       TofBinning =[300, -.001, 16666.7],
                       CalibrationParameters='DIFC' ,
                       PeakPositions=dvalues,
                      OutputCalibrationTable='new_cal',
                      DiagnosticWorkspaces='diagnostics')
```

marshallmcdonnell, 2018-06-19:
After merging with master and rebuild, unable to also reproduce the issue. Closing as fixed

closed 2018-06-19 by marshallmcdonnell