mantidproject/mantid #22591: Instrument View issues on macOS High Sierra

gvardany, 2018-06-15:
Instrument view with `OpenGL` enabled suffers on macOS 10.13. 

### Expected behavior
Instrument viewer should open without errors.

### Actual behavior
Instrument viewer throws errors. Few examples below:

[image]
[image]

### Steps to reproduce the behavior
1. Install mantid nightly on macOS 10.13. 
2. LoadEmptyInstrument (any would do, for example TOPAZ).
3. Open the instrument view

### Platforms affected
macOS 10.13 (maybe also 10.12?), fine on 10.11 though

mantid-roman, 2018-06-15:
It could be helpful to know which gl command issued the initial error.

quantumsteve, 2018-06-15:
Have you tried building Mantid on 10.13? Does the same issue exist?

OwenArnold, 2018-06-18:
For information, i am running version 10.12.6 Sierra and have not been able to reproduce these issues.

LamarMoore, 2018-06-19:
@gvardany Have you monitored your GPU when loading the instrument? I noticed a similar error situation yesterday when I ran out of GPU memory.

gvardany, 2018-06-19:
Good points, I haven't monitored the GPU neither tried building on 10.13 yet. Will keep you posted.

gvardany, 2018-06-20:
I've just tried to build on 10.13 itself, and the problems are gone.

gvardany, 2020-02-17:
No longer relevant.

closed 2020-02-17 by gvardany