mantidproject/mantid #22596: TOFTOF GUI is leaking temporary workspaces

mganeva, 2018-06-18:
Reported by @soininen (PR #22589)
**Original reporter:** [username facility]/[nobody]. 

### Expected behavior
After the end of the data reduction all temporary workspaces should be deleted.

### Actual behavior
3 temporary workspaces remain after the data reduction is finished. If the data reduction will run again, 3 more temporary workspaces will appear an so on. 

### Steps to reproduce the behavior
The temporary workspaces appear after deleting the workspace `g{prefix}DataCleanFrame`
```python
DeleteWorkspaces([gnewDataCleanFrame])
```

### Platforms affected
all

soininen, 2018-06-18:
Maybe some of the workspaces getting deleted are still referenced in the Python script? I mean, there is a shared pointer to them in Python's side, maybe they are used or just remain alive until Python's garbage collector runs?

mganeva, 2018-06-18:
Good point, I will check that.

JoachimCoenen, 2018-06-26:
the leaked workspaces seem to be caused by the TOFTOFCropWorkspace algorithm. but copying the PyExec code directly into the script no ws are leaked. strange...

Leaks ws:
``` python
wsData1 = Load(Filename='27')
gwsDataCleanFrame1 = TOFTOFCropWorkspace(wsData1)
```

Does not leak ws:
``` python
# copied and modified from TOFTOFCropWorkspace:
def toftofCropWS(inputws, outputws):
    run = inputws.getRun()
    channel_width = float(run.getLogData('channel_width').value)
    full_channels = float(run.getLogData('full_channels').value)
    tof1 = float(run.getLogData('TOF1').value)
    outputws = CropWorkspace(inputws, XMin=0., XMax=full_channels*channel_width + tof1, OutputWorkspace=outputws)
    return outputws
    
wsData1 = Load(Filename='27')
gwsDataCleanFrame1 = toftofCropWS(wsData1, 'gwsDataCleanFrame1')
```

soininen, 2018-06-26:
I think the problem with `TOFTOFCropWorkspace` is that it is using the same name for the *OutputWorkspace* of `CropWorkspace` as it's own *OutputWorkspace*. You could try avoiding the ADS in the call to `CropWorkspace`. Maybe this would work:
```python
class TOFTOFCropWorkspace(PythonAlgorithm):
    [...]
    def PyExec(self):
        """ Main execution body
        """
        inputws = self.getProperty("InputWorkspace").value

        run = inputws.getRun()
        channel_width = float(run.getLogData('channel_width').value)
        full_channels = float(run.getLogData('full_channels').value)
        tof1 = float(run.getLogData('TOF1').value)

        outputws = api.CropWorkspace(inputws, XMin=0., XMax=full_channels*channel_width + tof1, StoreInADS=False)
        self.setProperty("OutputWorkspace", outputws)
```

mganeva, 2018-06-28:
@soininen Antti, thanks a lot. I've tried it on my mac. This fixes the problem.

@JoachimCoenen Could you please commit this change (file `TOFTOFCropWorkspace.py`, line 74, add `StoreInADS=False`)? Thanks!