mantidproject/mantid #22651: CreateWorkspace algorithm is extremely slow for large workspace dimensions

gcassella, 2018-06-22:
Using CreateWorkspace to create large workspaces can take significantly longer than workarounds to achieve a similar effect. For example,

```python
x = np.random.random((10000, 801))
y = np.random.random((10000, 800))
e = np.random.random((10000, 800))

ws = CreateWorkspace(DataX=x, DataY=y, DataE=e, NSpec=10000)
```

requires significantly longer to run than 

```python
ws = CloneWorkspace(some_dummy_workspace_with_NSpec)

for i in range(10000):
   ws.setX(i, x[i, :])
   ws.setY(i, y[i, :])
   ws.setE(i, e[i, :])
```
On ISIS compute I have seen speed differences of up to 60x between these two methods of filling a large workspace with arbitrary data.

NickDraper, 2018-08-02:
@OwenArnold did you say that @nvaytet was looking into this?

nvaytet, 2018-08-03:
It is probably due to writing the full data arrays as a string to the workspace history. See #20443 and #22664. One option is to use an external library (boost karma or Abseil, as proposed by Steven Hahn), and the other is to use OpenMP threads to speedup the creation of the long strings.

closed 2018-08-20 by gcassella