mantidproject/mantid #22673: Multiple fitting muon label

AnthonyLim23, 2018-06-26:
The label for the simultaneous fitting is not updating correctly when the run changes (in the muon GUI).

If the workspace already exists it should cancel the fit and not append the name

NickDraper, 2018-07-10:
@AnthonyLim23 Is this a duplicate of #22824 ?

AnthonyLim23, 2018-07-10:
ah no. This is different. So here if the user has created a custom name (say bob) then does a second fit it should ask if the user wants to overwrite. If the user clicks no the fit should be cancelled. 

Issue #22824 is the name becoming rubbish under certain circumstances.

NickDraper, 2018-07-13:
@AnthonyLim23 What is the state of this issue?
Is there a PR, Is it vital for this release?

AnthonyLim23, 2018-07-13:
this can be next release. Its more of a new feature than a bug fix

closed 2019-07-19 by AnthonyLim23