mantidproject/mantid #22710: GEM texture focusing, mismatch between number of vanadium banks and cropping values

josephframsay, 2018-06-29:
**Original reporter:** Winfried Kockelmann, ISIS

**Note this is a high priority for ISIS Powder for 3.14**

The new texture capabilities in `isis_powder` for GEM, while seeming to run fine in testing, failed as soon as we tried to show them to GEM scientists. Specifically, when running `create_vanadium` we got the mismatch error from `common.crop_banks_using_crop_list`, as `isis_powder` provides 160 cropping values (in texture mode), but for some reason the vanadium was focused into 162.

Though Winfried sent me his cycle mapping file, I couldn't reproduce the error on my machine, and I suspect that the failed run wasn't using the right vanadium. We should take a cycle mapping file which we know to work with texture mode up there and see if that fails.
```
83881-84212:
  label: "cycle_17_3"
  offset_file_name: "offsets_2011_cycle111b.cal"
  Rietveld:
    vanadium_run_numbers: "83883" #VNb
    empty_run_numbers: "83884"
  PDF:    
    vanadium_run_numbers: "83883" #VNb
    empty_run_numbers: "83884"
```

A few ideas:
- I suspect the issue may be that something secret and evil in `DiffractionFocussing` is just discarding any banks that don't have data in, and the vanadium run we're using just happens to have the right number thrown out that we end up with 160
- The grouping file contains 164 banks. This is weird, as the whole point is that we focus into 160, so try and investigate why this is (this will be hard, as it was made by a sandwich student in days of yore, and scientists don't seem to remember much about it). It's possible that we may need to create another grouping file which does this properly (I don't think it will come to this)

josephframsay, 2018-06-29:
Assigned @DavidFair, but not really - it's just so that this gets passed on instead of getting lost in the issue graveyard, as it's an important one

josephframsay, 2018-06-29:
Take a look at #22610 - I don't know how closely this relates to this issue but it certainly seems relevant!

marshallmcdonnell, 2018-06-29:
It sounds exactly like what is happening to us. We find it when an entire focussed group's pixels are masked, you get back less spectrum then expected. 

From the review in #22610, I am trying to refactor the algorithm to clean it up and take this behavior out altogether. For a quick fix, you can use that branch if you like but it does not yet have the option exposed to `AlignAndFocusPowder` so it is only in `DiffractionFocussing`. 

With the refactor I am working on now with the default to not do this, it will propagate to both `AlignAndFocusPowder` and `AlignAndFocusPowderFromFiles`

Hope that helps.

SamJenkins1, 2019-03-28:
I don't believe this is an Issue any longer, Texture mode focusing has been working for scientists while other issues in it were being tested.

closed 2019-03-28 by SamJenkins1