mantidproject/mantid #22716: Optimisation level for Release is not the same as RelWithDeb

DavidFair, 2018-06-29:
The optimisation level for Release is `-O3` whilst RelWithDeb is `-O2`. 

This potentially makes profiling less applicable as a lot of the overhead / improvements from vectorisation is no longer visible.

A small change the the CXX flags in the CMake files would resolve this

martyngigg, 2018-07-02:
@DavidFair CMake sets the default flags for the different modes. I suspect they use `-O2` for `RelWithDebInfo` so that the debugging information is of some use within the debugger. For the default modes I would prefer it if we stick with what cmake provides to avoid confusion with documentation for example.

DavidFair, 2018-07-05:
Okay, closing as this is set by CMake rather than us as pointed out by Martyn

closed 2018-07-05 by DavidFair