mantidproject/mantid #22733: Project recovery obscures logs

peterfpeterson, 2018-07-02:
When users run mantid with logging at a level of "information," logs of the algorithms for the project recovery are obscuring the logs from the algorithms they are actively running. Also, it was unclear if the user was seeing the progress bar moving related to this.

This is related to the umbrella of #22729.

martyngigg, 2018-07-03:
I don't see the logging at information but I do see the progress bar kick up occasionally. I can fix that.

DavidFair, 2018-07-03:
I followed up with @TWJubb 

With a number of workspaces and their logger set to debug they see a large number of messages which say:

```
Determining ioRaw buffer size
Unable to find loadraw.readbuffer.size in the properties file
loadraw.readbuffer.size not found, setting to 200000
```

This excessive logging consumes a CPU for large periods of time, even at notice level. This would result in a drop in performance across Mantid whenever people have a number of workspaces open, so is definetly worth looking at as part of this. (If this wasn't the original issue)

martyngigg, 2018-07-03:
I can see the problem with the excessive logging and think I can put a solution in that can improve it.