mantidproject/mantid #22735: New reflectometry loader for future Nexus files

reimundILL, 2018-07-03:
Nexus files will be updated for the next cycle starting in September requiring an update of the loader.

This will be a second version of the loader:
- a third version of Nexus entries would be too difficult to handle for the existing loader
- a simplification of the loader makes sense and is in progress (https://github.com/mantidproject/mantid/issues/22734) -> the loader should have two inputs which are `Filename` and `OutputWorkspace`

reimundILL, 2018-07-03:
About the input options: it may be good to add `XUnits` and `ExtractMonitors` enabling a quick visualization of the data in unit wavelength

soininen, 2018-07-03:
The new algorithm should probably move the detector to what is currently returned by `LoadILLReflectometry::sampleDetectorDistance()` and implement the angle options in `LoadILLReflectometry::detectorRotation()`, except the one that uses the direct beam as a reference:
* Read the detector angle and possible deflection angle from NeXus. There could be an option to use the sample angle as well, similarly to SAN option in COSMOS.
* A user-given angle (*BraggAngle*) could be used to override the detector/sample angle.

closed 2021-03-02 by stale[bot]