mantidproject/mantid #22751: Unscripted Testing ISIS Indirect interfaces

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See
* http://www.mantidproject.org/Indirect:ConvertToEnergy
* http://www.mantidproject.org/Indirect:Indirect_Data_Analysis
* http://www.mantidproject.org/UsingElwin
* http://www.mantidproject.org/UsingFury

AnthonyLim23, 2018-07-09:
The documentation needs updating: 
#22822 
#22820 
#22819 
#22818

AnthonyLim23, 2018-07-09:
for Reduction I got the error

```Error in execution of algorithm ISISIndirectEnergyTransfer:
No registered converter was able to produce a C++ rvalue of type class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > from this Python object of type NoneType
  at line 257 in 'C:/Users/BTR75544/nightly/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\ISISIndirectEnergyTransfer.py'
  caused by line 590 in 'C:/Users/BTR75544/nightly/MantidInstall/scripts/Inelastic\IndirectReductionCommon.py'
  caused by line 670 in 'C:/Users/BTR75544/nightly/MantidInstall/scripts/Inelastic\IndirectReductionCommon.py'
Unknown error starting next batch algorithm
Got error from algorithm "ISISIndirectEnergyTransfer"
Stopping batch algorithm because of execution error
```

AnthonyLim23, 2018-07-09:
For data  data analysis:

I can only do a colour fill plot of the _elf ws
The other two loo different to the example in Elwin

Running MSDFit - when I try to drag and drop the workspace I get an unhandled exception 
Trying to move the blue line causes an exception 
Inface trying to do anything causes an exception

AnthonyLim23, 2018-07-10:
The bug in the reduction (Above is still present). @brandonhewer  do you know about this?

AnthonyLim23, 2018-07-10:
In data analysis:

The plot difference tick box in the fit browser seems to be ignored (so we should hide it - this is not a bug)

In conv fit the fit browser does not update with the fit values

@brandonhewer what is the equivalent of fury?

brandonhewer, 2018-07-10:
There is no clear equivalent of FuryFit.

brandonhewer, 2018-07-10:
The reason colour fill plot cannot be done on the other workspaces in Elwin is as a result of there only being a single spectrum. The ELF workspace is transposed (Q-values are on the y-axis) -- so the colour fill plot option is available.

brandonhewer, 2018-07-10:
The issue here in ISIS Energy Transfer seems to be that you are using Custom Detector Grouping with an empty string -- there seems to be no added validation for this -- so we're getting a rather strange error down the line.

NickDraper, 2018-07-13:
@AnthonyLim23 Is this complete now?  If so please close the issue

closed 2018-07-13 by NickDraper