mantidproject/mantid #22755: Unscripted Testing DGSReduction

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See http://www.mantidproject.org/Direct:DgsReduction
     and http://www.mantidproject.org/Create_MD_Workspace_GUI#Direct_Inelastic_Mod_Q

AndreiSavici, 2018-07-09:
Works on HYS. Fails to find monitor peaks on SEQ (event monitors). Not used at SNS.
@mganeva is testing it for TOF-TOF

mganeva, 2018-07-10:
Works for TOFTOF. Testted on mantid nightly for mac os x.

closed 2018-07-10 by AndreiSavici