mantidproject/mantid #22761: Unscripted Testing TOFConverter

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See http://docs.mantidproject.org/v3.12.0/interfaces/TOF%20Converter.html

gemmaguest, 2018-07-05:
Testing on Ubuntu 16.04

gemmaguest, 2018-07-06:
Tested on Ubuntu 16.04

No major problems found. Results look correct, sensible error messages given, correct widgets enabled/disabled for specific conversions.

The only very minor issue I found is a slightly misleading validation message when converting e.g. Wavelength to Q. The validation says that Theta > 0 is required, but if you enter Theta==0 then it does not give an error but produces the result Q=0.

closed 2018-07-06 by gemmaguest