mantidproject/mantid #22762: Unscripted Testing ISIS SANS

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See http://www.mantidproject.org/SANS_Data_Analysis_at_ISIS

KarlPalmen, 2018-07-09:
Installing Mantid nightly.

KarlPalmen, 2018-07-09:
Testing Mantid 3.12.20180703.1554 .

KarlPalmen, 2018-07-09:
Testing on Windows 7.

KarlPalmen, 2018-07-09:
The button to load the User File was called "Load Mask" and the only such file in the loqdemo folder was called MaskFile.txt . This took me some time to figure out. Documentation  referred to in test instructionsmay need improving.

KarlPalmen, 2018-07-09:
The user file produced the same settings early on, but leter ones were different and I could not identify where the range 0.13 to 0.25 would be. I've been unable to do basic reduction with this and error messages are not helpful.

KarlPalmen, 2018-07-09:
I've found out that despite the error messages the reduction was successful and produced the workspaces described. Unlike the muon interface, plotting is not automatically done.

KarlPalmen, 2018-07-09:
It seem to work OK, but did crash when playing around with it roughly. I used the report facility to report it under name "Karl #22762" .

KarlPalmen, 2018-07-09:
I tried getting it to crash again and it did not.

KarlPalmen, 2018-07-09:
It seems OK for me, except that the issue of confusion between mask files and user files needs addressing.

KarlPalmen, 2018-07-09:
I found out that the documentation is in the wiki, so I made a small change to say the the user file can be reffered to as a mask file.

closed 2018-07-09 by KarlPalmen