mantidproject/mantid #22765: Unscripted Testing ORNL SANS

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

NickDraper, 2018-07-17:
The time for this is fast running out.  We will have to close it, untested if it has not been completed, at the end of Wednesday.

ricleal, 2018-07-17:
All good from the ORNL :shipit:

closed 2018-07-17 by ricleal