mantidproject/mantid #22767: Unscripted Testing FilterEvents

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

AndreiSavici, 2018-07-09:
I've looked at `HYS_13657` from systemtests and filtered on `s1`. When I try to plot the s1 value, I get ```  File "/home/andrei/Mantid/mantid/scripts/FilterEvents/eventFilterGUI.py", line 713, in plotLogValue
    vectimes.append(tf)
AttributeError: 'numpy.ndarray' object has no attribute 'append'
```

AndreiSavici, 2018-07-09:
The abobe issue is fixed in PR #22838

Mantid-Matthew, 2018-07-10:
I am testing in Ubuntu 16.04

Mantid-Matthew, 2018-07-10:
I found that certain workspaces would not load in the event filter I have created an issue to cover this #22862

AndreiSavici, 2018-07-12:
We should check if the array type is float or int. `M3` is a string array. One can check at https://github.com/mantidproject/mantid/blob/release-next/scripts/FilterEvents/eventFilterGUI.py#L811 if the first value is a real number. Change that line to `if (len(times) > 1) and numpy.isreal(p.value[0]):`

Mantid-Matthew, 2018-07-12:
There are now PR's addressing all the issues I found.

closed 2018-07-12 by Mantid-Matthew