mantidproject/mantid #22769: Unscripted Testing Vates

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Including data loading in Vates via ParaView. For help see [[Testing_Notes_2.5.0|here]]

NickDraper, 2018-07-13:
@OwenArnold @LamarMoore Could one of you give vates in the release a once over today please.  I don't expect any issues as I'm not aware of any work in the VSI, but we have ran into unexpected problems in the past before anyway.

We have done some work in the sliceviewer to make peaks work with non-orthogonal axes.

OwenArnold, 2018-07-13:
Yes, I'll pick this up.

OwenArnold, 2018-07-13:
Aside from https://github.com/mantidproject/mantid/issues/22914, SliceViewer and VSI seem to be working. Tested on OSX 10.12

closed 2018-07-13 by OwenArnold