mantidproject/mantid #22770: Unscripted Testing Engineering Diffraction

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See http://docs.mantidproject.org/v3.8.0/interfaces/Engineering_Diffraction.html

Mantid-Matthew, 2018-07-04:
Testing on Ubuntu 16.04

Mantid-Matthew, 2018-07-04:
In the focus window I entered 74044 in the run number window for crop focusing and entered 7 in the focusing window all the buttons in the GUI were disabled and the windows refused to close. It is possible this just kicked off a long running background process but nothing in MantidPlot seemed to indicate this.

Mantid-Matthew, 2018-07-06:
As a usability point when attempting to run a calibration in the Make new calibrations section the GUI appears to search for the relevant Vanadium and Calibration runs behind the scenes and will claim the input boxes are empty until it has found them.

Mantid-Matthew, 2018-07-06:
Other than my various issues with the archive search I didn't find any significant problems here. I have created #22806 to address this.

closed 2018-07-06 by Mantid-Matthew