mantidproject/mantid #22773: Unscripted Testing ORNL Powder Diffraction

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See http://www.mantidproject.org/PowderDiffractionReduction

wdzhou, 2018-07-16:
On REL 7, tests were done for both POWGEN and VULCAN data.  

There are two issues that have been found.  Issue #22950 will be resolved in this release.  Issue #22951 will be resolved in the next release.

closed 2018-07-17 by NickDraper

wdzhou, 2018-07-17:
Unscripted testing ORNL powder reduction passed.  Issue #22951 is resolved and closed by PR #22977.