mantidproject/mantid #22774: Unscripted Testing Sample Transmission calculator

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See http://docs.mantidproject.org/v3.7.2/interfaces/SampleTransmissionCalculator.html

ElliotAOram, 2018-07-11:
| Area | Issue |
|------|-------|
| Input wavelength - single | |
| Input wavelength - multiple | This lets you have massive rebinning boundaries that go unvalidated such as `-99999999999999999999999, 0.01, 99999999999999999999999`. Running this will take forever and results in a bad allocation error at the end of it. |

ElliotAOram, 2018-07-11:
@keeeto I don't think either of these are release stopping but might be worth keeping track of them somewhere?

closed 2018-07-11 by ElliotAOram