mantidproject/mantid #22776: Unscripted Testing ISIS SANS (v2 GUI)

keeeto, 2018-07-04:
You have been assigned unscripted testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to unscripted testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://www.mantidproject.org/SANS_Data_Analysis_at_ISIS

NickDraper, 2018-07-11:
@MikeHart85 Will you have a chance to look at this in the next day or so?

MikeHart85, 2018-07-11:
@NickDraper No, I'm afraid I have my hands full at the moment.

gemmaguest, 2018-07-13:
Testing on Ubuntu 16.04

gemmaguest, 2018-07-13:
One minor issue is that when a file can't be found the error message reports the row as a 0-based index (e.g. "there was a bad entry for row 1" means the second row, not the first). This would probably be less confusion for users as a 1-based index.

NickDraper, 2018-07-13:
@Mantid-Matthew what do you think?

Mantid-Matthew, 2018-07-13:
@NickDraper I think these all look like relatively easy fixes which if it's ok with you I might try and get in for the release ?

PR #22947 (2018-07-17): Sans usability fixes
**Description of work.**
This PR tidies up some minor issues with the ISIS SANS GUI

**To test:**

* Check that the tab ordering in the GUI is sensible. 
* Check that the wavelength slice and event slice tooltips make sense.

Fixes #22776. 

Does this update require release notes?
- [ ] Yes
- [X] No

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidp [... cut]
closed 2018-07-18 by NickDraper