mantidproject/mantid #22815: LoadNexusMonitors, LoadOnlyMonitors should load histograms if both are present

NickDraper, 2018-07-06:
### Expected behavior
this does not work as the file contains both histogram and event monitors
```` python
LoadEventNexus(Filename='LET00006278.nxs', OutputWorkspace='LET00006278', 
LoadMonitors=True)
```

### Actual behavior
this works
``` python
LoadEventNexus(Filename='LET00006278.nxs', OutputWorkspace='LET00006278', 
LoadMonitors=True, MonitorsLoadOnly="Events")
```

but we should have the default be to warn, not error, and then load histogram monitors is no option is selected.

### Steps to reproduce the behavior

### Platforms affected
All

NickDraper, 2018-07-06:
High priority as this will affect all LET files.

peterfpeterson, 2018-07-06:
I had it error if you ask for monitors and there either aren't any or they are ambiguous. The old behavior was event worse as `LoadNexusMonitors` would tend towards histogram and `LoadEventNexus(LoadMonitors=True, ...)` would tend towards events. Any change should still error out if there aren't monitors and the user tried to load them.

NickDraper, 2018-07-09:
This change will only affect the behaviour if both event and histogram monitors are present.  We need the default Load to work by default, otherwise occasional uses will just give up and complain that Mantid does not work.
Good point about the documentation, I'll do that.

closed 2018-07-10 by NickDraper