mantidproject/mantid #22816: Update SANS Documentation

Mantid-Matthew, 2018-07-06:
A lot of the current sans documentation is out of date and still on the old wiki. This should gradually be updated revalidated with scientists and moved to the new sphinx documentation.
**Original reporter:** stephen.king@stfc.ac.uk

Mantid-Matthew, 2018-07-06:
Haveing asked around the wiki does have one advantage which is that scientists can edit the pages themselves. I therefore need to speak to the relevant scientists before deciding how to go ahead with this.

DavidFair, 2019-09-16:
I spoke to Steve (@smk78) regarding to issue and I'll quickly recap here the outcomes:

- The Mantid wiki for SANS is very outdated, referring only to the old GUI and obsolete information. This will be removed
- In it's place the documentation will move into .rst files so we can keep it up to date as updates progress. This reduces the burden on both ourselves and instrument scientists
- We greatly welcome any documentation they wish to contribute, and I've discussed three ways which come to mind:

1. Scientists who feel comfortable and want to use Sphinx and .rst with Github accounts will be allowed to open pull requests to contribute their documentation
2. They can type up documentation in an alternative format if they prefer and pass it onto the SANS team who will convert it over.
2. Alternatively they could submit an email or issue with some text they would like add to the documentation and that will also be added.

It's important to stress that the documentation moving into .rst files is not meant to be a barrier. We encourage feedback from a "user perspective" and hope this change will significantly reduce the barrier of maintaining a separate wiki section

PR #26854 (2019-10-02): Update ISIS Sans Documentation
**Description of work.**
Starts pulling in some of the docs from [the ISIS SANS Wiki](https://www.mantidproject.org/ISIS_SANS) into .rst files

- Split ISIS SANS Interfaces docs into their own folder
- Chunk down each page
- Update all the images, and organise them into a folder
- Some text fixing, however maybe 1/4 was proofed - the rest will be do as the docs are updated
- Fixed some other minor docs issues (such as pages not appearing in categories)

There is some additional work to port from the wiki, but these are more edge case pages which each need to be p [... cut]
closed 2019-10-02 by gemmaguest