mantidproject/mantid #22845: Compton Gaussian Profile in fitting causes a hard crash

keeeto, 2018-07-09:
This crash case existed in Mantid 3.12. It is related to allowing any type of function to be used in conjunction with the plot guess option

### Expected behavior

Mantid should be able to handle Gaussian Compton Profile. 

### Actual behavior

Mantid crashes in the Indirect Interface is Gaussian Compton Profile is added

### Steps to reproduce the behavior

Go to Indirect> Data Analysis > ConvFit

* In ConvFit load the sample and resolution provided below
* Choose Fit Type to be One Lorentzian
* Check Plot Guess
* Choose Add Function
* Choose General > Gaussian Compton Profile
* Mantid crashes

[QENSFitTestFiles.zip](https://github.com/mantidproject/mantid/files/2177045/QENSFitTestFiles.zip)

### Platforms affected

All

SpencerHowells, 2018-07-09:
Gaussian Compton Profile is not a QENS Function it is for Vesuvio. It should therefore NEVER be used in the QENS Interface. For a Gaussian peak the simple Gaussian Function should be used. How do you propose to stop users selecting totally inappropriate functions? Should we?

keeeto, 2018-07-09:
@SpencerHowells. Fair enough point. I think that we could just block these functions completely. As it stands a user could mistakenly add the function and bring the whole code crashing down; we certainly need to stop this happening!!

SpencerHowells, 2018-07-09:
@keeeto There are dozens of functions in the library that could be selected. How do we decide which ones should be blocked? Or do we check all the ones we think are appropriate and risk leaving many out. Then when we think of a new one do we need to edit the code or do we create a library of suitable ones which has to be updated. As far as I am concerned if an user is stupid enough to chose a bad one so be it!! When all else fails read the manual!!!! Or discuss with a instrument scientist. Also they could create their own function & crash the system.

keeeto, 2018-07-10:
@SpencerHowells I know that there are many functions there. I respectfully disagree with the option of let them suffer for their mistakes. I can see scenarios where the slip of a mouse causes the incorrect function to be chosen, causing a crash in Mantid; that's not a good outcome. If we are able to prevent Mantid from hard crashing due to simple user mistakes then we should do it wherever possible.

SpencerHowells, 2018-07-10:
@keeeto I've loaded 1 of our QENS data sets & tried using the Fit wizard with just the GaussianComptonProfile function. When I try to enter a Mass value it comes up with exception error. The log gives 
Unexpected exception: ComptonProfile - Unable to find DetectorGroup component parameter "t0"
t0 I think is the 'time zero' which is used by Vesuvio but not QENS.
With MultiDataSet interface when I try to add the function it crashed.
So it is not ConvFit. 
Using an inappropriate function is NOT a simple error!  I rest my case.

closed 2019-03-13 by NickDraper