mantidproject/mantid #22858: Indirect data reduction fails

AnthonyLim23, 2018-07-10:
If you follow the instructions at http://www.mantidproject.org/Indirect:ConvertToEnergy then you will get error messages in the logger:

```No registered converter was able to produce a C++ rvalue of type class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > from this Python object of type NoneType
  at line 257 in 'C:/Users/BTR75544/nightly/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\ISISIndirectEnergyTransfer.py'
  caused by line 590 in 'C:/Users/BTR75544/nightly/MantidInstall/scripts/Inelastic\IndirectReductionCommon.py'
  caused by line 670 in 'C:/Users/BTR75544/nightly/MantidInstall/scripts/Inelastic\IndirectReductionCommon.py'
Unknown error starting next batch algorithm
Got error from algorithm "ISISIndirectEnergyTransfer"
Stopping batch algorithm because of execution error```

ElliotAOram, 2018-07-11:
@brandonhewer this failure is seen as a result of the `Custom` grouping mode being the default option in the gui. 
If left blank this then gets converted to `None` by the `_str_to_none` function in IndirectCommon. 
It is then passed to the grouping algorithm. Which does not expect the `None` type.
For fixing this, is it safe to assume if the custom option is left blank then we should just do no grouping (e.g. in this case `0-50`)

SpencerHowells, 2018-07-11:
@ElliotAOram @brandonhewer I think 0-50 default is the best

SpencerHowells, 2018-07-11:
@ElliotAOram We’ve got 0-50 wrong! It’s the min/max detector number as given in the gui. In this case 3-53.

NickDraper, 2018-07-13:
Sam, can you pick this up from elliot for today

PR #22891 (2018-07-13): ISIS Indirect data reduction: Change default grouping option and add validation
**Description of work.**

* Added validation to the Indirect data reduction gui to ensure custom grouping is not empty
* Changed the default grouping to be individual meaning users don't have to input into that field unless they physically change the detector grouping option.

**Report to:** @brandonhewer 

**To test:**

* Open mantid plot
* Open the Indirect data reduction interface (Interfaces > Indirect > Data Reduction)
* Observe that in the `Detector Grouping` section, the mode should default to `Individual`
* [... cut]
PR #22952 (2018-07-17): Indirect Data reduction use the same syntax as GroupDetector algorithm
**Description of work.**

This is an additional part of the fix for #22858. The original PR stopped the crash behaviour however, without this PR, the custom grouping option in the interface does not perform any of the grouping operations as expected. This is a small change that will allow users to use the same syntax for the group detectors algorithm in the custom grouping field.

**To test:**

* Open mantid plot
* Open the Indirect data reduction interface (Interfaces > Indirect > Data Reduction)
* Enter the run number 213 [... cut]
closed 2018-07-13 by NickDraper