mantidproject/mantid #22859: Indirect fit not updating

AnthonyLim23, 2018-07-10:
In the indirect data analysis if you follow these instructions 

http://www.mantidproject.org/Indirect:Indirect_Data_Analysis

you will find that the fit property browser does not update after a fit. This can be misleading.

SpencerHowells, 2018-07-10:
These appear to be for the old interface not the new one in the release

AnthonyLim23, 2018-07-10:
The instructions still work for the new interface and the bug is present in the new interface.

brandonhewer, 2018-07-10:
I can't seem to replicate -- can you retry on the current release branch and give the instructions which led to no updating.

AnthonyLim23, 2018-07-11:
I did do it on the release branch (after your PR got merged, the one Keith looked at). I just followed the instructions in the link.

AnthonyLim23, 2018-07-11:
I think that I have found the problem now. It is in general fitting, not sure if it is intentional or not, if you do a fit and it does not converge the values are not updated in the fit browser. For example see https://github.com/mantidproject/fitbenchmarking/blob/master/benchmark_problems/Neutron_data/WISH17701_peak9.txt with the simplex minimizer. @mantid-roman is this the intended behaviour?

keeeto, 2018-07-11:
I guess that this issue should probably be closed - there could be an issue around general fitting, but not for the Elwin interface in particular.

closed 2018-07-11 by keeeto

mantid-roman, 2018-07-11:
I cannot reproduce it in general fitting