mantidproject/mantid #22883: Project recovery causes offspec reduction to crash

gemmaguest, 2018-07-11:
**Original reporter:** Jos at ISIS

### Expected behavior
Project recovery should not cause Mantid to crash

### Actual behavior
Crashes while executing GeneratePythonScript algorithm when saving workspace histories. It seems to crash reliably in the nightly build from 3 July. I've reproduced it once on Windows 7 in a debug build at commit 0be626b951 to get the stack trace below but it's intermittent. I can't reproduce it on Ubuntu.

```
 	MantidKernel.dll!std::_Fetch_add_relaxed_4(volatile unsigned long * _Tgt, unsigned long _Value) Line 1583	C++
 	MantidKernel.dll!std::_Atomic_fetch_add_4(volatile unsigned long * _Tgt, unsigned long _Value, std::memory_order _Order) Line 1606	C++
 	MantidKernel.dll!std::atomic_fetch_add_explicit(std::_Atomic_int * _Atom, int _Value, std::memory_order _Order) Line 976	C++
 	MantidKernel.dll!std::_Atomic_int::fetch_add(int _Value, std::memory_order _Order) Line 1196	C++
 	MantidKernel.dll!boost::detail::atomic_increment(std::atomic<int> * pw) Line 31	C++
 	MantidKernel.dll!boost::detail::sp_counted_base::add_ref_copy() Line 98	C++
 	MantidKernel.dll!boost::detail::shared_count::shared_count(const boost::detail::shared_count & r) Line 485	C++
 	MantidKernel.dll!boost::shared_ptr<Mantid::Kernel::PropertyManager>::shared_ptr<Mantid::Kernel::PropertyManager>(const boost::shared_ptr<Mantid::Kernel::PropertyManager> & r) Line 405	C++
 	MantidKernel.dll!boost::shared_ptr<Mantid::Kernel::PropertyManager>::operator=(const boost::shared_ptr<Mantid::Kernel::PropertyManager> & r) Line 526	C++
 	MantidKernel.dll!Mantid::Kernel::PropertyManagerOwner::operator=(const Mantid::Kernel::PropertyManagerOwner & po) Line 31	C++
 	MantidKernel.dll!Mantid::Kernel::PropertyManagerOwner::copyPropertiesFrom(const Mantid::Kernel::PropertyManagerOwner & po) Line 84	C++
 	MantidAPI.dll!Mantid::API::Algorithm::initializeFromProxy(const Mantid::API::AlgorithmProxy & proxy) Line 934	C++
 	MantidAPI.dll!Mantid::API::AlgorithmProxy::createConcreteAlg(bool initOnly) Line 246	C++
 	MantidAPI.dll!Mantid::API::AlgorithmProxy::execute() Line 65	C++
>	MantidPlot.exe!MantidQt::ProjectRecovery::saveWsHistories(const Poco::Path & historyDestFolder) Line 458	C++
 	MantidPlot.exe!MantidQt::ProjectRecovery::projectSavingThread() Line 384	C++
 	MantidPlot.exe!MantidQt::ProjectRecovery::projectSavingThreadWrapper() Line 341	C++
 	MantidPlot.exe!MantidQt::ProjectRecovery::createBackgroundThread::__l2::<lambda>() Line 227	C++
 	[External Code]	
```

### Steps to reproduce the behavior
- Add these scripts to the python directories path: ***to follow***
- Set default instrument to OFFSPEC
- Enable ISIS data archive

### Platforms affected
Windows 7

gemmaguest, 2018-07-11:
@DavidFair This also results in the problem I showed you where we have a recovery script with missing history for hidden workspaces.

NickDraper, 2018-07-11:
@keeeto Could you look into reproducing this

DavidFair, 2018-07-11:
Hmm at a quick glance at the stack trace it appears the workspace which the shared pointer refers to is deleted underneath us.

I don't believe the ADS deletes shared pointers (else we would have seen this already). One convoluted way this could be happening is:

- An algorithm takes a copy of the shared pointer
- It removes it from the ADS, so believes it has the only handle
- It then removes / alters the workspace underneath in a way we can't handle
- When we try to dereference the same handle we except

One way to check is to break in the debugger when we throw and look at the shared pointers use count. This is an atomic field on the type which could hint (if there are more than 1 owners) someone else also is holding the pointer and modifying it.

gemmaguest, 2018-07-11:
I'm just building the latest release branch to see if it's still an issue. I'll have a go when it's finished.

gemmaguest, 2018-07-11:
I can't reproduce the issue on the current release branch (commit 6b587ba0) but it may just be working by luck

gemmaguest, 2018-07-11:
I got a different stack trace this time:
```
 	MantidAPI.dll!Mantid::API::Algorithm::processGroups() Line 1337	C++
 	MantidAPI.dll!Mantid::API::Algorithm::doCallProcessGroups(Mantid::Types::Core::DateAndTime & startTime) Line 1262	C++
 	MantidAPI.dll!Mantid::API::Algorithm::execute() Line 558	C++
 	MantidAPI.dll!Mantid::API::AlgorithmProxy::execute() Line 67	C++
>	MantidPlot.exe!MantidQt::ProjectRecovery::saveWsHistories(const Poco::Path & historyDestFolder) Line 458	C++
 	MantidPlot.exe!MantidQt::ProjectRecovery::projectSavingThread() Line 384	C++
 	MantidPlot.exe!MantidQt::ProjectRecovery::projectSavingThreadWrapper() Line 341	C++
 	MantidPlot.exe!MantidQt::ProjectRecovery::createBackgroundThread::__l2::<lambda>() Line 227	C++
 	[External Code]	
```

The AlgorithmProxy looks corrupt - it was created with the name `GeneratePythonScript` but m_name is `CloneWorkspace` and m_alg is null.

gemmaguest, 2018-07-12:
No, I can't reproduce it on the HEAD of the release branch

gemmaguest, 2018-07-12:
@martyngigg I applied your commit 849ab67 from #22733 to the version that was crashing and I haven't been able to reproduce the problem since, so I'm reasonably confident this has fixed it.

closed 2018-07-12 by gemmaguest

martyngigg, 2018-07-12:
@gemmaguest Brilliant. I'm glad that did the trick. Thanks for confirming.