mantidproject/mantid #22917: Running setCell on a TableWorkspace does not update an open view in MantidPlot

samueljackson92, 2018-07-13:
**Original reporter:** Lewis Edwards (ISIS)

### Expected behavior
Running `setCell` on a table workspace that has an open view in MantidPlot should update the data in the view.

### Actual behavior
Running `setCell` on a table workspace that has an open view in MantidPlot does not change the data in the view.

### Steps to reproduce the behavior
Here's a script to reproduce the issue
```python
from mantid.api import WorkspaceFactory
tbl = CreateEmptyTableWorkspace()

tbl.addColumn("double", "l")
tbl.addColumn("double", "k")
tbl.addColumn("double", "h")

tbl.addRow([1,1,1])

# Go open table and view data

tbl.setCell(0,0, 100)
```

### Platforms affected
All

Pasarus, 2018-07-16:
Attempt to do it for other methods including addColumn etc.

Pasarus, 2018-07-16:
Make a list of things that need updating and added, for modifying etc.

Potentially add ITableWorkspace from not python exports' method Modified in Workspaces.

Pasarus, 2018-07-17:
List of things to add modified to:

-  All of the Workspace classes via inheritance from Workspace.class
- ITableWorkspace::setCell
- ITableWorkspace::addColumnPlotType
- Load new workspace?
- ITableWorkspace::addRowFromSequence
- ITableWorkspace::addRowFromDict
- ITableWorkspace::addColumnSimple
- ITableWorkspace::setValue instead of setCell? or both?
- ITableWorkspace::setPlotType
- Workspace::setTitle
- Workspace::setName
- WorkspaceGroup::addWorkspace
- WorkspaceGroup::removeAll
- WorkspaceGroup::removeItem
- WorkspaceGroup::add
- WorkspaceGroup::remove
- IMDWorkspace::setConvention
- IMDWorkspace::setMDMasking
- MatrixWorkspace::initialize (const std::size_t &NVectors, const std::size_t &XLength, const std::size_t &YLength)
- MatrixWorkspace::initialize (const std::size_t &NVectors, const HistogramData::Histogram &histogram)
- MatrixWorkspace::initialize (const Indexing::IndexInfo &indexInfo, const HistogramData::Histogram &histogram)
- MatrixWorkspace::maskBin
- MatrixWorkspace::operator=
- MatrixWorkspace::replaceAxis
- MatrixWorkspace::setBinEdges
- MatrixWorkspace::setCounts
- MatrixWorkspace::setCountStandardDeviations
- MatrixWorkspace::setCountVariances
- MatrixWorkspace::setDistribution
- MatrixWorkspace::setFrequencies
- MatrixWorkspace::setFrequencyStandardDeviations
- MatrixWorkspace::setFrequencyVariances
- MatrxiWorkspace::setHistogram
- MatrixWorkspace::setImageE ---- Follow this down to where it's overridden
- MatrixWorkspace::setImageY ---- Follow this down to where it's overriden
- MatrixWorkspace::setIndexInfo
- MatrixWorkspace::setMaskedBins
- MatrixWorkspace::setMDMasking
- MatrixWorkspace::setMonitorWorkspace ---- Follow down to where it is overridden
- MatrixWorkspace::setPoints
- MatrixWorkspace::setPointStandardDeviation
- MatrixWorkspace::setPointVariances
- MatrixWorkspace::setSharedDx
- MatrixWorkspace::setSharedE
- MatrixWorkspace::setSharedX
- MatrixWorkspace::setSharedY
- MatrixWorkspace::setTitle
- MatrixWorkspace::setX (const std::size_t index, const Kernel::cow_ptr< HistogramData::HistogramX > &X) ---- Deprecated
- MatrixWorkspace::setX (const std::size_t index, const boost::shared_ptr< HistogramData::HistogramX > &X) ---- Deprecated
- MatrixWorkspace::setYUnit
- MatrixWorkspace::setYUnitLabel
- MatrixWorkspace::updateSpectraUsing
- MatrixWorkspace::setX

samueljackson92, 2018-07-17:
As there are so many, perhaps it would be better to have a new issue for this work as it is no longer specific to ITableWorkspace.

Pasarus, 2018-07-17:
Alright I will make a new issue and assign it to myself with regards to this.

PR #22961 (2018-08-17): ITableWorkspace now updates GUI when data changed via Python
**Description of work.**
Made sure that setCell, addColumn, setPlotType and addRow when used in the script interpreter or the script window, update any open windows, by utilizing ITableWorkspace::modified()

**Report to:** Lewis Edwards (ISIS) 

**To test:**

- Create empty table:
```python
from mantid.api import WorkspaceFactory
tbl = CreateEmptyTableWorkspace()

tbl.addColumn("double", "l")
tbl.addColumn("double", "k")
tbl.addColumn("double", "h")

tbl.addRow([1,1,1])
```
- Open the table as a window first.
- Then Test using the co [... cut]
closed 2018-08-17 by peterfpeterson