mantidproject/mantid #22950: Remove 'Send to' cluster in reduction UI

wdzhou, 2018-07-16:
**Original reporter:** [username facility]/[nobody]. 

### Expected behavior

*Send to* Fermi is no longer used in reduction UI.  Remove it and related tab.

### Actual behavior

### Steps to reproduce the behavior

### Platforms affected

NickDraper, 2018-07-17:
For this to make the release it needs to be complete, tested and merged by the end of Wednesday.

Are you confident that removing this unused functionality will not have any side effects on the rest of the interface?  Is removing it important for this release.  If it is just about reducing maintenance of the code should it be done on master rather than release-next?

peterfpeterson, 2018-07-17:
The systems that it connects to haven't existed for a few years. The tab was discovered during unscripted testing.

PR #22977 (2018-07-18): Remove "Send cluster" in reduction UI
**Description of work.**

Remove *Send cluster* and related classes and methods in SNS reduction framework.

**To test:**

Launch MantidPlot on analysis cluster.  Check DGS reduction, Powder reduction and SANS reduction UIs.

Fixes #22950. 

Does this update require release notes?
- [ ] Yes
- [ ] No

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](h [... cut]