mantidproject/mantid #22963: LoadLog skips logs

mantid-roman, 2018-07-17:
**Original reporter:** Freddie Akeroyd

### Expected behavior
LoadLog should load all logs from the log file.

### Actual behavior
Sometimes it skips some logs.

### Steps to reproduce the behavior
Load the same workspace from two files:
1. \\\\isis\inst$\ndxoffspec\instrument\data\cycle_17_1\OFFSPEC00042712.raw
2. \\\\isis\inst$\ndxoffspec\instrument\data\cycle_17_1\OFFSPEC00042712.nxs

The workspace loaded from the nexus file has more logs than that from the raw file.

### Platforms affected
All

FreddieAkeroyd, 2018-07-17:
Hi Roman, I took a bit more of a look and it seems to be due to the fact they have a block called NSPECTRA and a log called this is already created by the raw loader. The current LoadLogs() implementation seems to exit when it gets an error rather than continuing with the rest of the blocks.  Moving the exception handling to inside the loops at the end of loadThreeColumnLogFile() seems to fix it.

mantid-roman, 2018-07-17:
Thanks, Freddie.

FreddieAkeroyd, 2018-07-17:
Shall i push a fix to a branch?

mantid-roman, 2018-07-17:
Yes, please.

FreddieAkeroyd, 2018-07-17:
Ok, I've pushed a fix. I know a release is happening shortly and we had a problem recently on OFFSPEC with some logs not making it into the NeXus file so I was keen this workaround option was working. Thanks.

mantid-roman, 2018-07-17:
The branch should be based on `release-next` not `master`. I'll make a new branch and a new PR.

FreddieAkeroyd, 2018-07-17:
Sorry, I didn't read down far enough to see release-next. I'm a bit out of practice at Mantid development! Did "rebase --onto" not work then?

FreddieAkeroyd, 2018-07-17:
good idea - thanks

PR #22976 (2018-07-18): Allow loading of logs after duplicate detected
**Report to:** Freddie Akeroyd

**Description of work.**

Move the exception handling to inside the loops so that log
loading continues after e.g. a duplicate is detected. An issue
was spotted on OFFSPEC where a block called NSPECTRA had been
created and this conflicted with a sample log created by the
LoadRaw routine itself.

**To test:**

Load raw file for OFFSPEC run 42712 which will call LoadLogs but you will not see sample logs such as 
s1hc and s1vc which are visible in the corresponding nxs file. After merging PR you should see s1hc and s1v [... cut]
closed 2018-07-18 by martyngigg