mantidproject/mantid #22990: Input file finder fails to find files with uppercase extension ".RAW"

nvaytet, 2018-07-18:
### Expected behavior

Mantid should find files with extensions `.RAW` and `.raw`.

### Actual behavior

It seems that it is only looking for `.raw` files.

Trying the `ConvertToEnergy` tutorial in the Mantid examples, it is not possible to run the reduction by clicking the `Run` button because the input file is not found:
```
findRuns hint = IRIS21360
vector findRun('IRIS21360', exts[0])
Looking for IRIS at ISIS.
Instrument 'IRIS' found as IRIS at ISIS.
Add facility extensions defined in the Facility.xml file
getExtension(IRIS21360, exts[7])
Failed to find extension. Just using last '.'
Looking for IRIS at ISIS.
Instrument 'IRIS' found as IRIS at ISIS.
getFullPath(IRS21360.nxs)
Searching for IRS21360.nxs in /home/nvaytet/work/mantid/Sample_Datasets/SampleData-ISIS/
getFullPath(IRS21360.raw)
Searching for IRS21360.raw in /home/nvaytet/work/mantid/Sample_Datasets/SampleData-ISIS/
getFullPath(IRS21360.sav)
Searching for IRS21360.sav in /home/nvaytet/work/mantid/Sample_Datasets/SampleData-ISIS/
getFullPath(IRS21360.add)
Searching for IRS21360.add in /home/nvaytet/work/mantid/Sample_Datasets/SampleData-ISIS/
getFullPath(IRS21360.nxspe)
Searching for IRS21360.nxspe in /home/nvaytet/work/mantid/Sample_Datasets/SampleData-ISIS/
Unable to find run with hint IRIS21360
Unable to find file path for IRIS21360
```

This does not only apply to the `Indirect Data Reduction` interface, the same behaviour is observed by entering `IRS21360` in the main Mantid `Load Dialog` (above the workspace list).

### Steps to reproduce the behavior

Follow the "Basic Reduction" steps in the [ConvertToEnergy](http://www.mantidproject.org/Indirect:ConvertToEnergy) tutorial in the Mantid examples.

### Platforms affected

Only tested Ubuntu 18.04, but this may apply to others?

NickDraper, 2018-07-30:
linux based OS's are case sensitive by default, whereas windows are not, which is why all of these things work on windows.

There is an open question as to whether Mantid should be case insensitive on linux based OS's as that would differ from what is expected in that OS.

However in this case the extension is hidden and provided by the mantid internals.

NickDraper, 2018-07-30:
It appears that the only options are raw, and RAW according to the data files we have, and we do not need to support Raw, rAw or raW etc.

This we can do most easily by extending the facilities file, but that might have a cost in performance for times when no file can be found

NickDraper, 2018-07-30:
We do have a .properties flag that can set Linux based OS's to be case insensitive, that would solve this.

```
filefinder.casesensitive
```

The current default is off, which is probably correct.

closed 2018-07-30 by NickDraper