mantidproject/mantid #23018: Crash doing Show Detectors on result of ExtractPolarizationEfficiencies

gemmaguest, 2018-07-19:
### Expected behavior
Mantid should not crash. I'm not sure if Show Detectors menu option should be available for this workspace

### Actual behavior
Mantid crashes

### Steps to reproduce the behavior
- Set up mantid to use the parameters file from #22997 and start mantid
- Run this script:
```
Load(Filename='OFFSPEC00044913', OutputWorkspace='OFFSPEC00044913')
ExtractPolarizationEfficiencies(InputWorkspace='OFFSPEC00044913_1', OutputWorkspace='efficiencies_1')
```
Right-click on `efficiences_1` and select Show Detectors. Mantid crashes.

### Platforms affected
Ubuntu 16.04, probably all

gemmaguest, 2018-07-19:
I get similar behaviour with `LoadILLPolarizationFactors` in v3.12.1. Run the [usage example](http://docs.mantidproject.org/nightly/algorithms/LoadILLPolarizationFactors-v1.html#usage), right-click on `pol_factors` and Show Detectors and mantid crashes.

mantid-roman, 2018-07-19:
Any workspace without attached instrument will make mantid crash

gemmaguest, 2018-07-19:
Ah, yes. `CreateWorkspace` followed by Show Detectors crashes too.

AndreiSavici, 2018-07-19:
Merged into release-next

PR #23020 (2018-07-19): Crash doing Show Detectors on a workspace without an instrument
**Description of work.**
Check if the instrument has the sample. If it doesn't do not attempt to create the table. Show a warning.

**To test:**
* Create a workspace
```
ws = CreateWorkspace([0, 1], [1, 1])
```
* Try to open the detector table.

Fixes #23018

Does this update require release notes?
- [ ] Yes
- [X] No

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does  [... cut]
closed 2018-07-19 by AndreiSavici