mantidproject/mantid #23025: Crash when opening BinMD dialog in MantidPlot

OwenArnold, 2018-07-20:
```python

Load(Filename='SXD23767.raw', OutputWorkspace='SXD23767', LoadMonitors='Exclude')
ConvertToDiffractionMDWorkspace(InputWorkspace='SXD23767', OutputDimensions='Q (lab frame)',
                                              SplitThreshold=50, LorentzCorrection='1',
                                              MaxRecursionDepth='13', Extents='-15,15,-15,15,-15,15')                                     
```

Open in `BinMD` and get crash below. Did open a second time though without crashing?

```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libMantidQtWidgetsCommonQt4.dylib	0x0000000107b7c701 MantidQt::MantidWidgets::SlicingAlgorithmDialog::hasDimensionHistoryChanged() const + 449
1   libMantidQtWidgetsCommonQt4.dylib	0x0000000107b7b9f2 MantidQt::MantidWidgets::SlicingAlgorithmDialog::buildDimensionInputs(bool) + 114
2   libMantidQtWidgetsCommonQt4.dylib	0x0000000107b7b12d MantidQt::MantidWidgets::SlicingAlgorithmDialog::initLayout() + 1709
3   libMantidQtWidgetsCommonQt4.dylib	0x000000010798f163 MantidQt::API::AlgorithmDialog::initializeLayout() + 483
4   libMantidQtWidgetsCommonQt4.dylib	0x00000001079b0aca MantidQt::API::InterfaceManager::createDialog(boost::shared_ptr<Mantid::API::IAlgorithm>, QWidget*, bool, QHash<QString, QString> const&, QString const&, QStringList const&, QStringList const&) + 874
5   MantidPlot                    	0x0000000106c4109c MantidUI::createAlgorithmDialog(boost::shared_ptr<Mantid::API::IAlgorithm>) + 1100
```
Possibly related to [this](https://github.com/mantidproject/mantid/commit/af234963a3c05064d16cb1752fb6ab3fe5b89a4b#diff-9fec5927f4cd5736cf9f8a36b623ba78) change by @samueljackson92 ?

NickDraper, 2018-07-20:
I've tried to reproduce this on windows a couple of times and I've not managed yet ...

NickDraper, 2018-07-20:
Is it possible you had some data in your algorithm history that caused the issue?