mantidproject/mantid #23026: Unable to Debug on unsupported developer platform Lubuntu

SamJenkins1, 2018-07-20:
### Expected behavior
Debug build of mantid builds correctly
### Actual behavior
build cannot find some libpoco files and therefore errors out
### Steps to reproduce the behavior
use cmake to create a debug version and attempt to build it on lubuntu
### Platforms affected
lubuntu

SamJenkins1, 2018-07-20:
its easy enough to fix this on an individual computer by simply manually installing the missing files, though it does need to be the 31-dbg versions of them.

NickDraper, 2018-07-30:
Why do we care about Lubuntu?

closed 2018-07-30 by NickDraper

DavidFair, 2018-07-30:
This should be fixed in #23050 

Also Lubuntu should function equivalently to Ubuntu, as it's purely the display environment which differs. So I suspect this was a case that @SamJenkins1 was using an older version of Mantid developer package where we weren't correctly pulling in the `libpoco-*dbg` packages