mantidproject/mantid #23027: ISIS Reflectometry Interface - Runs Tab Tidying

edwardb96, 2018-07-20:
*This issue refers to work which should be based on changes currently only present in the branch `isis_reflectometry_alternative_table`.*

**Main Tasks**
- [ ] Build an interface around the `runPythonScript` method, `PythonRunner`, implement it for
      `QtReflMainWindowView` and pass in the instance when creating the `ReflRunsTabPresenter`. This
      seems like the cleanest way to enable running python code without having all presenters know
      about the parent view and it's ability to run python code. I believe a technique like this is
      necessary for plotting until Mantid 4, but we may be able to avoid it completely for `CatalogLoginDialog`,
      this would require some further investigation and potentially hacking in `AlgorithmDialog`. 
      *Estimated time*: 1 day for first suggestion using `PythonRunner` interface implemented by `QtReflMainWindowView`.
      *Required for*: Plotting, ICat Login.
- [ ] Add methods for accessing the status of the generate notebook flag and the jobs from the `RunsTable`. 
    *Estimated time*: Less than an hour.
    *Required for*: Processing, Autoprocessing.
- [ ] I'm not sure it's worth trying to MVPify the search section, this seems to have been kept reasonably
      separate, some of the ICat stuff in `ReflRunsTabPresenter` should be moved out into it's own
      class.
    *Estimated time*: Assuming search section is left alone, a couple of hours, requires further investigation.
    *Required for*: Autoprocessing.
- [ ] Currently, for autoprocessing we handle polling/diffing the list of runs explicitly in
      `RunsTabPresenter` this makes it more difficult to mock out and test the arrival of new runs.
      If we move this out into a new class, `CatalogRunNotifier` an implementation of a new
      interface `RunNotifier` which is injected into the presenter either as a parameter (or if
      necessary a factory which takes the session details as a parameter) then this becomes possible.
      This should probably also be the new home for the `metadataFromDescription` method.

- [ ] Remove the `getPresenter` method left behind from MVC.
- [ ] Move away from old style notify(Flag) interface to use of separate methods.
- [ ] Finally rename and move the files into correct directory structure, updating any include paths as necessary.

marshallmcdonnell, 2019-06-24:
Just thought I would suggest this for the Jupyter Notebook stuff: https://github.com/mwouts/jupytext

gemmaguest, 2019-06-27:
Closing this issue as remaining tasks have been consolidated into #26025

PR #26011 (2019-06-26): Reflectometry GUI: Change defaults for Instrument Settings
**Description of work.**
Change what the new Reflectometry GUI will have as the default for NormalizeByIntegratedMonitors and CorrectDetectors, to True from False.

**To test:**
- Open Workbench or MantidPlot
- Open Reflectometry->ISIS Reflectometry
- Switch to tab Instrument Settings
- NormalizeByIntegratedMonitors and CorrectDetectors should have ticked checkboxes

Fixes #23027

*This does not require release notes* because it fixes an issue with unreleased software

---

#### Reviewer ####

Please comment on the following ([full des [... cut]
closed 2019-06-27 by gemmaguest