mantidproject/mantid #23046: ISIS Reflectometry Interface - Make Default Value Loading Consistent across GUI and Algorithms

edwardb96, 2018-07-23:
*This issue refers to work which should be based on changes currently only present in the branch `isis_reflectometry_alternative_table`.*

There is some ambiguity over the expected behaviour in the event of empty/default values. Some
default's exist in different levels of the GUI, some exist in the algorithms. The behaviour should be unionised among all of these as much as possible. We should then be able to retrieve default values via an instance of the model (created from an algorithm instance) and injected into the presenter on construction.

The current suggested approach with changes to the GUI is somewhat reliant upon the interface being able to obtain a set of default values for all fields at the time when the interface is launched, this is because, in switching numeric fields to `QSpinBox`s it has become difficult for numeric inputs to be empty. If this constraint proves problematic we could move back to using `QLineEdits`, this shifts more custom validation back into the presenters though.

The current suggestion is to try to load all settings from the parameter file and show
an error, preventing usage if this is not possible for any non-optional settings. This has not yet
been cleared with all instrument scientists.

gemmaguest, 2018-12-12:
Most of the spin box values are optional so I've used Qt's facility to treat and show zero as Unset. I think this reduces the priority for this but it would still be good to do.

gemmaguest, 2019-05-14:
In the initial implementation in PR #24923 instrument and experiment tab defaults are considered separately and are unset if any error was found when populating the tab. This is reported in the log. This is consistent whether opening the GUI, changing the instrument or restoring defaults. We can adjust the exact behaviour in future if required following user feedback.

gemmaguest, 2019-06-25:
Closing this as the initial work is done

closed 2019-06-25 by gemmaguest