mantidproject/mantid #23091: ISIS Powder: Polaris Feedback

DavidFair, 2018-07-25:
Here is the feedback I received from Polaris regarding ISIS Powder

- [ ] File extensions are not being respected in Mantid E.g. `POL12345.s01` opens `POL12345.nxs`. This happens with Data search acrhive, does it happen using the filesystem directly?
- [ ] We need to get their current adv. config values and update Mantid to ship these optimal defaults instead
- [ ] Absorption corrections require the multiple scattering flag whilst running `focus`, despite multiple scattering not currently being available
- [ ] File extension parameter is applied to the vanadium and empty runs (we should use .nxs always)
- [ ] Check with Helen that rename workspace group works correctly in the latest release

Long Term:
- [ ] Check all parameters are valid before running the main scripts

SamJenkins1, 2018-09-20:
I assume the main scripts here refers to stuff like create_vanadium and focus?

samueljackson92, 2018-09-20:
I would assume that as well.

SamJenkins1, 2018-10-04:
It appears that the advanced config values in mantid are already up-to-date

closed 2019-05-29 by SamJenkins1