mantidproject/mantid #23092: ISIS Powder: Pearl Feedback

DavidFair, 2018-07-25:
Here is the feedback I received from Pearl regarding ISIS Powder:

High Priority - Preventing use in production:
- [ ] Vanadium spline has a tail where we proceed to divide by small values (this could be resolved by next point)
- [ ] Get 0 counts where we "bottom out", instead an offset should be applied to the dataset to ensure it cannot be <= 0. 

Normal priority:
- [ ] Long mode name does not get appended to the final group workspace name causing confusion
- [ ] Separate spline co-eff in adv. config for long mode
- [ ] `long_mode` does not emit a warning when it is changed in any way unlike other params

samueljackson92, 2018-07-25:
For the high priority issues do you have an example script & files that will reproduce the issue?

samueljackson92, 2018-07-25:
Also perhaps a screenshot of the problem is appropriate.

DavidFair, 2018-07-25:
Attached is the script which relies on files from Joe to avoid setup:
```python
from isis_powder.pearl import Pearl

pearl_focus = Pearl(user_name="Test",
                          calibration_directory=r"\\olympic\Babylon5\Public\JoeRamsay\PEARL\calib",
                          output_directory=r"\\olympic\Babylon5\Public\JoeRamsay\PEARL\output",
                          config_file=r"\\olympic\Babylon5\Public\JoeRamsay\PEARL\pearl_shared_config.yaml",
                          calibration_mapping_file=r"\\olympic\Babylon5\Public\JoeRamsay\PEARL\pearl_run_mapping.yaml")

pearl_focus.create_vanadium(run_in_cycle=99481)
```

However I'm not seeing it with the above run. We need to contact Nick and ask him for more details such as run number and that he is running the latest version vs an older nightly

SamJenkins1, 2018-09-06:
I assume the point about spline co-eff is referring to moving 
```
"monitor_spline_coefficient": 20,
"spline_coefficient": 60,
```
out of `general_params` and having two separate versions, one in `long_mode_off_params` and one in `long_mode_on_params`, however I assume that if we're separating them they should have different values?

SamJenkins1, 2018-09-17:
The high priority issue is thought to be caused by sudden drops at that can occur at either end of the vanadium, for an example of this run
```
from isis_powder import Pearl

config_file_path = r"/home/sjenkins/Documents/Isis Powder/PEARL/pearl_shared_config.yaml"
Pearl_example = Pearl(config_file=config_file_path,user_name="test")
Pearl_example.create_vanadium(run_in_cycle=101508)
```
with these [files](https://github.com/mantidproject/mantid/files/2388752/PEARL.zip) (you'll need to update the paths involved)

SamJenkins1, 2018-09-19:
further looking into the high priority, the sudden drops seem to first appear after the c++ algorithm MergeRuns is called when creating the vanadium, looking into this further, it seems specifically to happen during the rebin within MergeRuns

martyngigg, 2019-02-19:
@SamJenkins1 Is this one done now? Can it be closed?

SamJenkins1, 2019-02-19:
I left it open as the scientists hadn't got round to testing yet, I've had an email saying that they have finally got round to it, so yes I suppose it can now.

martyngigg, 2019-02-19:
Great stuff!

closed 2019-02-19 by martyngigg