mantidproject/mantid #23101: Workbench - issues accessing plot window from script window

ianbush, 2018-07-25:
Accessing the plots from the script editor, e.g. to close or hide plots, rename the window etc. cause crashes in the workbench on Windows.

[image]

### Steps to Reproduce
Make a figure, e.g. `plt.plot([1,2,3])` or `plt.figure()`. Any of the following commands cause the crash on Windows:

 * `plt.figure(1).canvas.window().setWindowTitle('0.5')`
 * `plt.gcf().canvas.window().hide()`
 * `plt.figure(1).canvas.window().repaint()`

The same problems occur in both the editor window and the IPython console.

### Platforms affected

Problems seem to occur on Windows, and not occur on Ubuntu 16.04.

ianbush, 2018-07-27:
@martyngigg Replacing the canvas with the following seems to work for any of the canvas operations. I am a bit concerned about just how ugly this is, essentially any time a new canvas is created all of its methods need to get wrapped. It still does not help with any calls to methods on the window obtained by `canvas.window()`.

```python
class ThreadAwareCanvasQT(FigureCanvasQTAgg):
    def __init__(self, figure):
        print('Making our own canvas')
        FigureCanvasQTAgg.__init__(self, figure)
        self.wrap_methods()

    def wrap_methods(self):
        for attribute in dir(self):
            if hasattr(getattr(self, attribute), '__call__') and '__' not in attribute:
                setattr(self, attribute, QAppThreadCall(getattr(self, attribute), attribute))
```

martyngigg, 2018-07-30:
@ianbush I think we may have no choice in the threading scenario despite how ugly it is.

For the window we have our own MainWindow class in the `WorkbenchFigureManager` module. Can we do something similar?

ianbush, 2018-07-30:
@martyngigg I have pushed some changes doing something similar for the window and toolbar. This would need careful testing if we go with this approach, it is hard to reliably get all the correct things to wrap.

I am not sure if there are other things that could generate this error, presumably if you can get to other widget items (buttons in the toolbar perhaps?) you could get the same errors.

martyngigg, 2018-07-30:
I equally feel quite uneasy about this not that it has had to move outside of the backend. The only problem is I can't think of another way to run the script code within the workbench without locking up the GUI that doesn't involve multiple processes. Any ideas?

ianbush, 2018-07-31:
I don't really have any well thought through ideas. I was wondering how this is handled when something is added to the ADS, presumably that needs some kind of thread moving to work?

Some ideas:
 * We could try and keep the figure windows and the workbench window on the same thread, but I am not sure if that helps.
 * Move the Python script execution to the current figure thread. This would be a bit weird, and I think lock up to the figure window?
 * The other thing we could do is nothing - this only causes an issue for Windows debug builds. I am not sure this is the right thing to do though, and will be mildly annoying when testing on Windows.

martyngigg, 2018-07-31:
So for the threading stuff we need to distinguish between things that paint to the screen and things that don't. The issue is that, for any GUI application not just Qt, the thread that created the initial `QApplication` object is the only one that is allowed to paint to the screen. I'm not 100% sure why this is the case but it seems to be widely accepted (https://stackoverflow.com/questions/16287163/how-to-paint-onto-qlabel-in-another-thread).

For the ADS we are just inserting a pointer to heap-allocated memory into a map that itself exists on the heap so no need to worry about where the object lives as all threads share the same heap.

So in our case here we only have issues any time a widget comes to try to paint something from a separate thread. This was initially why I was hoping we could deal with it completely within the backend.

closed 2021-03-02 by stale[bot]