mantidproject/mantid #23106: Make MaxEnt and its Unit Test use new workspace features instead of deprecated

KarlPalmen, 2018-07-26:
Maxent still uses deprecated features such as `MantidVec` instead of new ones such as `BinEdges`, `counts` and `HistogramX` , which are explained in http://docs.mantidproject.org/v3.9.2/concepts/HistogramData.html . This issue is to ensure that new features are used instead of deprecated ones in all the Maxent code.

When done, this may facilitate any change to make Maxent work with worspaces with non-uniform bins.

KarlPalmen, 2018-07-26:
This is can be done when no work is being done on either #21577 or #21578.

AnthonyLim23, 2020-09-17:
The usage for this alg is not that high

MaxEnt.v1 | 3.10 | 34
MaxEnt.v1 | 3.11 | 136
MaxEnt.v1 | 3.12 | 75
MaxEnt.v1 | 3.13 | 99
MaxEnt.v1 | 3.6 | 35
MaxEnt.v1 | 3.7 | 15
MaxEnt.v1 | 3.8 | 6
MaxEnt.v1 | 3.9 | 21
MaxEnt.v1 | 4.2 | 17
MaxEnt.v1 | 5.0 | 51

closed 2021-03-23 by stale[bot]