mantidproject/mantid #23128: Crystal Field calculation is slow

me2d09, 2018-07-30:
**Original reporter:** Petr Cermak.

### Expected behavior
I am running MantidNotebook and calculate crystal field. I use:
cf = CrystalField('Ce', 'C4v',  B20=a, B40=b, B44=c)
cf.getEigenvalues()
1000x execution of such calculation takes 5.87s. When I do the same with my own pure python suite [CrysFiPy](https://crysfipy.rtfd.io), it is finished in 2.94s (2x faster). I guess that the problem could be, that each call of getEigenvalues() writes following to stdout:
> Error in logging framework: Cannot open fileCrystalFieldEnergies-[Notice] CrystalFieldEnergies started
CrystalFieldEnergies-[Notice] CrystalFieldEnergies successful, Duration 0.00 seconds

How to suppress this Notice?

### Steps to reproduce the behavior
You can use attached iNotebooks.
[CF-speed.zip](https://github.com/mantidproject/mantid/files/2241020/CF-speed.zip)

### Platforms affected
Windows 10, Mantid 3.12.1, MantidPython 2.7.10
For CrysFiPy: Python 3.6.5 | Anaconda (Windows 10)

NickDraper, 2018-08-02:
We've just disabled file logging in the most recent development versions of Mantid.  That would get rid of the first line.
 Error in logging framework: Cannot open file

Any nightly build from now should include that.

To suppress the other messages you need to adjust the logging level for Mantid.  Those log at Notice level, so if you set it up to warning (4) they would be suppressed.

``` python
ConfigService.SetLogLevel(4)
```

NickDraper, 2018-08-02:
@brandonhewer @robertapplin This is worth discussing with Sanghamitra and Spencer to consider the priority

AnthonyLim23, 2020-09-18:
@mducle is it still slow?

mducle, 2020-09-18:
Running the script in the notebook (as a script file - see below), I get:

```
[dl11170@host-192-168-243-146 ~]$ time mantidpythonnightly cfspeed.py 
FrameworkManager-[Notice] Welcome to Mantid 5.0.20200916.1754
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
DownloadInstrument-[Notice] All instrument definitions up to date
This function is under development and can be changed/removed in the future
CrystalFieldEnergies-[Notice] CrystalFieldEnergies started
CrystalFieldEnergies-[Notice] CrystalFieldEnergies successful, Duration 0.00 seconds
...
Time: 2.051969528198242

real	0m5.330s
user	0m4.663s
sys	0m1.041s
```

So it still takes about 5s to run the script but half of that time is initialising Mantid. The CF calculations themselves take 2s. So I would say that this problem is sorted.

(Disabling the logging using `ConfigService.setLogLevel(4)` as Nick suggested gives a marginal improvement (to around 1.7s for just the CF calculations).

----------------------

The script is:

```python
import mantid
from CrystalField import CrystalField
import random
import sys
import time

#mantid.ConfigService.setLogLevel(4)

cf = CrystalField('Ce', 'C4v',  B20=1.2208, B40=-0.0021, B44=0.2555)
results = []
lim = 2

def getB():
    b20 = random.uniform(-lim,lim)
    b40 = random.uniform(-lim,lim)
    b44 = random.uniform(-lim,lim)
    return [b20,b40,b44]

def run(amount):
    e1 = 4.95
    e2 = 24.26
    delta = 0.2

    x = 0
    y = 0
    t0 = time.time()
    while y < amount:
        bs = getB()
        cf = CrystalField('Ce', 'C4v',  B20=bs[0], B40=bs[1], B44=bs[2])
        e = cf.getEigenvalues()
        y += 1
        if abs(e[2] - e1) < delta * e1 and abs(e[4] - e2) < delta * e2:
            print("{}, {}, {}, {}".format(e[2], e[4], y, bs))
            x+=1
            results.append(bs)
            sys.stdout.flush()
    print("Time: {}".format(time.time()-t0))

run(1000)
```

AnthonyLim23, 2020-09-18:
I will close the issue then. @mducle if it is a problem reopen it.

closed 2020-09-18 by AnthonyLim23